loco-3d/crocoddyl #1018: Adding environment obstacles and modeling contact with the robot

nrkumar93, 2021-12-21:
Hi! Thanks for this wonderful library. 

I am trying to design a contact-rich environment for a robot manipulator. After looking into Croccodyl and Pinocchio documentation, I still have these 2 questions.
- How do I add obstacles as shown in the Crocoddyl readme (like [this](https://i.imgur.com/RQR2Ovx.gif) and [this](https://i.imgur.com/kTW0ePh.gif))
- How do I enable contact models to obtain contact forces when the robot interacts with the obstacle?

Please point me to any relevant issue, documentation, or piece of code. 
 
I really appreciate the great work, thanks
Ram

cmastalli, 2021-12-22:
Hi Ram,

I thank you for showing interests in our work.
Below, I provide answers to both points:

**1. Handling obstacles**
`ResidualModelPairCollision` is a residual function that allows us to penalize obstacle avoidance.
You can find more details here: https://github.com/loco-3d/crocoddyl/blob/master/include/crocoddyl/multibody/residuals/pair-collision.hpp

You can use this residual together with an appropriate activation model to penalize collision with obstacles.
Unfortunately, I haven't have time to prepare an example that uses this residual function.
Definitely, it would be important to have it, but ATM I am lacking of time for doing it.

Finally, in future, I will deliver a solver that can handle this (and any kind of) inequality constraint.

**2. Enabling contact forces**
Crocoddyl relies on a predefined contact schedule (for efficiency purposes).
Given this predefine schedule, you can enable the computation of the forces derivatives as shown here:
https://github.com/loco-3d/crocoddyl/blob/master/bindings/python/crocoddyl/utils/quadruped.py#L457-L458

`True` tells Crocoddyl to enable this computation.

Let me know if this answer your questions.
If so, could you close this issue?

nrkumar93, 2021-12-23:
Hi Carlos, 

Thanks for the super quick response and the links, they are certainly helpful. 

However, I am specifically looking for the example code for the results in the paper on 
- iCub robot jumping on OBSTACLES
- quadruped jumping on OBSTACLES. 

I could not find them in the `examples` directory or elsewhere. 
Thank you so much!

cmastalli, 2021-12-29:
Unfortunately, I had not included these examples since you can constructed them from current ones.

closed 2021-12-29 by cmastalli

fukangl, 2024-01-24:
Hi @cmastalli,

I am facing the similar problem of how to add objects to crocoddyl and model the contact between the object and the environment, and the robot. Could you provide me with some reference materials or suggestions? Thank you very much!

fukangl, 2024-01-29:
Hi @cmastalli,

I am facing a similar issue. How can I add some obstacles like [this](https://i.imgur.com/RQR2Ovx.gif) and visualize them? The question might seem simple, but I really require some guidance. Thank you!

fukangl, 2024-01-30:
Now I could load extra URDF files and visualize them. The question is how to build the contact models between the object and the robot. Thanks!