loco-3d/crocoddyl #1019: Interfacing IpOpt from Crocoddyl

PepMS, 2021-12-28:
Hello, 

during my PhD there have been several situations where I missed the possibility that a Crocoddyl problem could be solved using a general NLP solver. Basically, I would've liked to take advantage of Crocoddyl's functionalities to create an optimal control problem (dynamical models, costs models, etc.) but solve the resulting NLP with a solver like IPOPT. This could be useful to add arbitrary constraints, to test newly developed solvers, etc.

During this past week, I decided to give this a try and I implemented a very simple C++ class that takes a Crocoddyl optimal control problem and adapts it to be solved with the C++ IpOpt interface. You can find the code in this repo: https://github.com/PepMS/croco_ipopt

I decided to keep this in a separate repo because of 2 reasons:

1. Didn't want to mess with the Croco CMake to find IpOpt
2. Croco takes too much to compile

This is an ongoing development. I'm posting this since I'm looking for a little bit of advice/help in the implementation of this. Below, I'll try to detail the status of this development:

- [This class](https://github.com/PepMS/croco_ipopt/blob/master/src/ipopt.cpp) is the equivalent of a solver in Crocoddyl. It contains an object of [this other class](https://github.com/PepMS/croco_ipopt/blob/master/src/multiple-shooting-nlp.cpp), which is the one overriding the methods where costs functions, constraints, and their corresponding derivatives are evaluated by IpOpt.
- I've included a [very simple python binding](https://github.com/PepMS/croco_ipopt/tree/master/bindings/python/ipopt_croco) for the solver class, so I can adapt Crocoddyl examples easily.
- There are a few examples that I've tried:
  - [quadrotor.py](https://github.com/PepMS/croco_ipopt/blob/master/examples/python/quadrotor.py), [unicycle.py](https://github.com/PepMS/croco_ipopt/blob/master/examples/python/unicycle.py) and [arm_manipulation.py](https://github.com/PepMS/croco_ipopt/blob/master/examples/python/arm_manipulation.py). They work. The solver converges and it outputs a similar solution to that of Crocoddyl.
  - [quadrotor_ubound.py](https://github.com/PepMS/croco_ipopt/blob/master/examples/python/quadrotor_ubound.py) and [double_pendulum.py](https://github.com/PepMS/croco_ipopt/blob/master/examples/python/double_pendulum.py). They don't work. The solver does not converge.
- I don't know why the two examples do not work. I don't know whether this is a matter of the implementation or that the solver simply does not find an optimum.

There's a specific implementation detail that I would like to know your (the Crocoddyl's team) opinion. This is related to the fact that IpOpt does not support its variables to be in the SE(3) manifold. Thus, I had to adapt the objective function, constraints, and derivatives evaluation. What I implemented is that the solver should find an increment from the given warm(cold) start. Thus, the variables associated with the quaternion lies in the tangent space. This fact also adds some extra derivatives that should be taken into account. Even though I checked this countless times, there might be a mistake here.

Kudos to those that are reading this line :). If you find this would be a nice feature to have in Crocoddyl, I could find some time to implement this. Also, I'd love some guidance and code checking from you, since probably I'm missing very basic rules that go against time-efficient code. Besides, I haven't used IpOpt before, so any experience to tune the solver will be highly appreciated (e.g. in the examples where the solver is not converging).

Cheers!

cmastalli, 2021-12-29:
Hi @PepMS,

As discussed in private, integrating Ipopt in Crocoddyl is an useful feature.
It would be better for us (or particular for me), if you do not work in a third repository.
Instead, you might create a *draft PR*, where I could easily provide feedback or support in some development.

Furthermore, working on Crocoddyl repo won't make the development much harder, actually quite the opposite.
1. The compilation time won't increase as this is not a template class, you'll need to compile the Ipopt stuff only. 
2. The cmake file needs to include (optionally) the Ipopt.

And indeed, it would be useful to create unittest code using our factory. This helps to find where are the errors, and properly develop CI checks since there are many things to be appeared in Crocoddyl.

PepMS, 2022-01-03:
I'm closing this issue after creating PR #1020

closed 2022-01-03 by PepMS