loco-3d/crocoddyl #1027: Enhance Continuous Integration & Testing

wxmerkt, 2022-01-06:
- [ ] Add configuration for multithreading (e.g. clang multithreading fails unittests, cf. #1026)
- [ ] Build and test in Debug and Release mode
- [ ] Add configuration for vectorisation
- [ ] Add dashboard to track benchmark performance over PRs/commits to devel

cmastalli, 2022-01-06:
Testing vectorization will require to compile all the dependencies (pinocchio, hpp-fcl, eigenpy, etc) with the same flags, if we want to run Python-based tests and examples.
How are you planning to do this?

wxmerkt, 2022-01-06:
Good point. I don't have the bandwidth to implement these items but wanted to keep track of them. They are a good starting point if someone is keen to pick them up or if we'd have some engineer bandwidth

cmastalli, 2022-01-06:
> Good point. I don't have the bandwidth to implement these items but wanted to keep track of them. They are a good starting point if someone is keen to pick them up or if we'd have some engineer bandwidth

We could definitely start with testing compilation only (if this is not enough).

wxmerkt, 2022-03-31:
Pinocchio on ROS-CI now has HPP-FCL included, so the collision tests are run as well

cmastalli, 2023-10-20:
I added a configuration for codegen using CONDA; however, it would be nice to have another one in gitlab-ci.