loco-3d/crocoddyl #1032: Extending DifferentialActionModelFreeFwdDynamics

EpicDuckPotato, 2022-01-14:
Is it possible for me to write a python class that extends DifferentialActionModelFreeFwdDynamics, and use that with IntegratedActionModelEuler? I tried writing such a class, but whenever I call calc, it seems to call the parent class's calc method.

```
import pinocchio as pin
import crocoddyl as croc
import numpy as np
from scipy.spatial.transform import Rotation as R

class DifferentialActionModelFreeFwdDynamicsRotatingReferenceFrame(croc.DifferentialActionModelFreeFwdDynamics):
   def __init__(self, state, actuation, costs, frame_w):
     super().__init__(state, actuation, costs)
     print('derived constructor') # This prints correctly

  def calc(self, data, x, u):
    super().calc(data, x, u)
    print('derived calc') # This does not print

  def calcDiff(self, data, x, u):
    super().calcDiff(data, x, u)
    print('derived calcDiff) # This does not print
```

EpicDuckPotato, 2022-01-14:
I ended up just writing my derived class in cpp and using Boost Python to access it in Python.

cmastalli, 2022-01-15:
It is not possible to inherit classes that are wrapped in Python.
This is a limitation of Python boost.

What you could do is to implement this class in Python and adds the extra bits you need.
I suggest to work like this, as doing scientific code in c++ is a very long process.

closed 2022-01-15 by cmastalli