loco-3d/crocoddyl #1038: HTML animation

Saeed-Mansouri, 2022-02-02:
`HTML(animateCartpole(ddp.xs).to_html5_video())` command in `cartpole_swing_up.py` does't show any animations on my system (Ubuntu 18.04). To save the animation, I have used `animateCartpole(ddp.xs).save('basic_animation2.mp4', fps=30, extra_args=['-vcodec', 'libx264'])` command and it works. Can you help me to resolve the issue of HTML animation?

cmastalli, 2022-02-02:
Your issue is not clear. This code works as it is!
Please follow the instructions in the notebook

closed 2022-02-02 by cmastalli

Saeed-Mansouri, 2022-02-02:
Should I see an animation after running cartpole_swing_up.py?

cmastalli, 2022-02-02:
The code is designed to be use with the Jupiter notebook, and this is file for teachers (not students).

Again, you should use the notebook and complete the tasks written there.

Saeed-Mansouri, 2022-02-02:
I added the following command into `cartpole_swing_up.py`:

`animateCartpole(ddp.xs).save('basic_animation2.mp4', fps=30, extra_args=['-vcodec', 'libx264'])
`
By running `cartpole_swing_up.py`, I could save a video of the animation. As you can see in the below:

https://user-images.githubusercontent.com/13035122/152224521-ee6c54e1-bae2-44a4-ba1f-9bc96c1906e2.mp4

So the codes works well and I don't need to improve any parts of `cartpole_swing_up.py`. However, I expects that `HTML(animateCartpole(ddp.xs).to_html5_video())` that is considered in `cartpole_swing_up.py` shows the animation. My question is why `HTML(animateCartpole(ddp.xs).to_html5_video())` doesn't show any animation while the codes works well.

cmastalli, 2022-02-03:
> My question is why `HTML(animateCartpole(ddp.xs).to_html5_video())` doesn't show any animation while the codes work well

Please see the previous answer.
#1044 allows us to display it in python.