loco-3d/crocoddyl #1042: Python-derived models cause a segmentation fault when compiled in multi-threading mode

wxmerkt, 2022-02-02:
When compiled with multi-threading enabled, deriving models in Python causes a segmentation fault unless `nthreads=1` is set explicitly before calling `solve()`.  This makes sense, but we should have a user-friendly way of not segfaulting.

Case to reproduce:
```python
import crocoddyl
import numpy as np
from crocoddyl.utils import UnicycleModelDerived

if __name__ == "__main__":
    N = 200  # number of nodes

    def createProblem(model):
        x0 = np.matrix([[1.], [0.], [0.]])
        runningModels = [model()] * N
        terminalModel = model()
        xs = [x0] * (N + 1)
        us = [np.matrix([[0.], [0.]])] * N

        problem = crocoddyl.ShootingProblem(x0, runningModels, terminalModel)
        return xs, us, problem

    # Crashes
    xs, us, problem = createProblem(UnicycleModelDerived)
    problem.nthreads = 2
    # Fine as long as nthreads=1
    #xs, us, problem = createProblem(crocoddyl.ActionModelUnicycle)
    # problem.nthreads = 1
    solver = crocoddyl.SolverBoxDDP(problem)
    solver.solve()
```

cmastalli, 2022-02-02:
I am aware of this issue from a long time, but without a clear solution.

To be more concrete, Crocoddyl doesn't yet support multi-threading in python-derived code (action, residuals, cost models, etc).

It seems we need to manage the Python GIL using boost python. Honestly, I have tried different solutions, but still not clear how to do it.

If someone has a better understanding on how tackling this issue, then please go ahead or comment here.

wxmerkt, 2022-02-02:
I'm aware that this is a limitation and a "nice to have" to be solved. However, as long as it segfaults, it's a bug. If we can simply show an error message instead, it's fine to leave it as an enhancement. I don't have a solution for it right now though either :-/

cmastalli, 2022-02-02:
> I'm aware that this is a limitation and a "nice to have" to be solved. However, as long as it segfaults, it's a bug. If we can simply show an error message instead, it's fine to leave it as an enhancement. I don't have a solution for it right now though either :-/

Agreed. We could sort out the segfault. One solution is to change (behind the scene) the number of threads to 1 when Python-derived code is in place. Do you have another proposition?

cmastalli, 2022-02-03:
I think this post is relevant for anyone reading this thread: https://stackoverflow.com/questions/18618333/boost-python-wrap-functions-to-release-the-gil

closed 2022-02-03 by cmastalli