loco-3d/crocoddyl #1052: ActionModelNumDiff does not use the control bounds defined in model

quimortiz, 2022-03-17:
Minor: it seems that ActionModelNumDiff does not use the control bounds defined in the model that you pass in the constructor (member variable model_ in c++ code). Is that intended? Thanks for the library!

cmastalli, 2022-03-17:
Hi @quimortiz!

We use the action numdiff for unittesting only. That's the reason why we haven't added this support.

But I see the benefit for users to add such kind of feature. I will include it asap.

Is this fine for you?

quimortiz, 2022-03-17:
Thanks! Yes sure, it is not urgent. For now, I am defining the bounds on the ActionModelNumDiff object.

wxmerkt, 2022-03-18:
Isn't this fixed in `devel` by https://github.com/loco-3d/crocoddyl/pull/1048?

cmastalli, 2022-03-18:
You're right @wxmerkt!

@quimortiz -- we haven't released this code, but you could get it if you compile the devel branch.

I will close this issue!

closed 2022-03-18 by cmastalli