loco-3d/crocoddyl #1057: Enhance build instructions

justinberi, 2022-03-28:
Description:

I had difficulty building from source due to dependency versions. I was able to build from source using robotpkg to install Pinocchio, example-robot-data and gepetto-viewer. Though this approach was not documented and I haven't locked down versions. 

Do you have any CI scripts or superbuilds or Dockerfiles to use as a reference to build a development environment?

I found https://github.com/Gepetto/gepetto-utils/tree/master/docker/crocoddyl but this seems out of date?

Context:
I develop in a Docker container, via VS Code Remote-Containers, to ensure isolation from my host.

devcontainer.json
```
{
	"name": "Ubuntu",
	"build": {
		"dockerfile": "Dockerfile",
		// Update 'VARIANT' to pick an Ubuntu version: hirsute, focal, bionic
		// Use hirsute or bionic on local arm64/Apple Silicon.
		"args": { "VARIANT": "focal" }
	},

	// Set *default* container specific settings.json values on container create.
	"settings": {},

	// Add the IDs of extensions you want installed when the container is created.
	"extensions": [],

    "runArgs": [
		"--network=host",
        "--gpus=all",
        "--privileged"
	],
    "containerEnv": { "DISPLAY": "${localEnv:DISPLAY}"},

	// Use 'forwardPorts' to make a list of ports inside the container available locally.
	// "forwardPorts": [],

	// Use 'postCreateCommand' to run commands after the container is created.
	// "postCreateCommand": "uname -a",

	// Comment out to connect as root instead. More info: https://aka.ms/vscode-remote/containers/non-root.
	"remoteUser": "vscode"
}
```

Dockerfile
```
# Configurations for the containers provided by vscode
ARG VARIANT=focal
ARG USERNAME=vscode
FROM mcr.microsoft.com/vscode/devcontainers/base:${VARIANT}
# Renew the ARG https://docs.docker.com/engine/reference/builder/#understand-how-arg-and-from-interact
ARG USERNAME

# A robotpkg approach
# https://stack-of-tasks.github.io/pinocchio/download.html

# Install dependencies for building
RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y install --no-install-recommends \
        build-essential \
        cmake \
        curl \
        git \
        time \
        lsb-release \
        gnupg2 \
        libglu1-mesa mesa-utils '^libxcb.*-dev' libx11-xcb-dev libglu1-mesa-dev libxrender-dev libxi-dev libxkbcommon-dev libxkbcommon-x11-dev \
    && rm -rf /var/lib/apt/lists/*

# Install crocoddyl dependencies from robotpkg
RUN echo "deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub $(lsb_release -cs) robotpkg" | tee /etc/apt/sources.list.d/robotpkg.list \
    && curl http://robotpkg.openrobots.org/packages/debian/robotpkg.key | apt-key add - \
    && apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y install --no-install-recommends \
        robotpkg-py38-example-robot-data \
        robotpkg-py38-pinocchio \
        robotpkg-py38-qt5-gepetto-viewer-corba \
        robotpkg-py38-qt5-gepetto-viewer \
    && rm -rf /var/lib/apt/lists/*
# Add robotpkg paths to the non-root user
# Add the install PYTHONPATH of crocoddyl
RUN echo "export PATH=/opt/openrobots/bin:${PATH}" >> /home/${USERNAME}/.bashrc \
    && echo "export PKG_CONFIG_PATH=/opt/openrobots/lib/pkgconfig:${PKG_CONFIG_PATH}" >> /home/${USERNAME}/.bashrc \
    && echo "export LD_LIBRARY_PATH=/opt/openrobots/lib:${LD_LIBRARY_PATH}" >> /home/${USERNAME}/.bashrc \
    && echo "export PYTHONPATH=/opt/openrobots/lib/python3.8/site-packages:/usr/local/lib/python3/dist-packages:${PYTHONPATH}" >> /home/${USERNAME}/.bashrc \
    && echo "export CMAKE_PREFIX_PATH=/opt/openrobots:${CMAKE_PREFIX_PATH}" >> /home/${USERNAME}/.bashrc

# More dev packages
RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y install --no-install-recommends \
        python3-matplotlib \
        python3-scipy \
        python3-tk
    && rm -rf /var/lib/apt/lists/*
```

cmastalli, 2022-03-28:
> I had difficulty building from source due to dependency versions

Could you tell us more about it? Otherwise, we cannot help you in this PR.

> Do you have any CI scripts or superbuilds or Dockerfiles to use as a reference to build a development environment?
> 
> I found https://github.com/Gepetto/gepetto-utils/tree/master/docker/crocoddyl but this seems out of date?

I am not sure if this file is out of date. @nim65s is the person you might want to ask.

nim65s, 2022-03-28:
Hi @justinberi,

Dockerfiles can be very different from one another depending on your goal, so it's hard to maintain a "standard" reference for that. 
In your particular case, I think the best I can do is provide a review. Starting from your Dockerfile, I went to this one:

```Dockerfile
ARG VARIANT=focal
FROM mcr.microsoft.com/vscode/devcontainers/base:${VARIANT}
ARG USERNAME=vscode
ARG VARIANT

ENV CMAKE_PREFIX_PATH=/opt/openrobots PATH=/opt/openrobots/bin:$PATH

ADD http://robotpkg.openrobots.org/packages/debian/robotpkg.key /

RUN --mount=type=cache,sharing=locked,target=/var/cache/apt \
    --mount=type=cache,sharing=locked,target=/var/lib/apt \
    apt-key add /robotpkg.key \
 && echo "deb [arch=amd64] http://robotpkg.openrobots.org/packages/debian/pub ${VARIANT} robotpkg" > /etc/apt/sources.list.d/robotpkg.list \
 && echo "/opt/openrobots/lib/python3.8/site-packages" > /usr/lib/python3/dist-packages/robotpkg.pth \
 && apt-get update -y \
 && DEBIAN_FRONTEND=noninteractive apt-get install -qqy \
    cmake \
    git \
    python-is-python3 \
    python3-matplotlib \
    python3-scipy \
    robotpkg-py38-example-robot-data \
    robotpkg-py38-qt5-gepetto-viewer-corba

# This last part below is just here to test everything, I guess you don't need it.

WORKDIR /src
ARG VERSION=1.9.0
ADD https://github.com/loco-3d/crocoddyl/releases/download/v${VERSION}/crocoddyl-${VERSION}.tar.gz .
RUN tar xf crocoddyl-${VERSION}.tar.gz \
 && cmake -B build -S crocoddyl-${VERSION} \
 && cmake --build build -j8 \
 && cmake --build build -t test \
 && cmake --install build \
 && rm -rf /src/*
 ```
 
If you need version pinning, this is doable with apt, but I guess conda would be the most convenient approach here.

And for build instructions, well, from my point of view this is just "use cmake", ie., in the canonical form we can see everywhere on the internet:
```
mkdir build
cd build
cmake ..
make
make test
make install
```

or in the modern form with `--build` / `--install` I used above. CMake should find everything it needs on your system, or tell you very explicitely otherwise. This should include required version numbers when applicable. Did you experience any issue with that ?

Another possibility is to use conda, which will take care of downloading / compiling the dependencies (which can be pinned) it needs behind your back, without the need for you to bother about that.

justinberi, 2022-03-29:
@cmastalli sorry for the confusion, I will reformulate my question below for clarity.

**Are there any standardised build recipes for a development environment?**

I think @nim65s has answered that with:

> Dockerfiles can be very different from one another depending on your goal, so it's hard to maintain a "standard" reference for that.

@nim65s thanks for the review of the dockerfile it was very useful. I may attempt to strip the Dockerfile dependencies down in the future. If I have the time to clean it up would you like me to provide it back as a method of creating a development environment?

Closing as this was a request for information which has been answered.

closed 2022-03-29 by justinberi