loco-3d/crocoddyl #1079: Core dumped

spykspeigel, 2022-07-04:
For the last few days I am unable to run the crocoddyl codes.
I am running in conda env with python 3.8.13, ubuntu 20.04. I did a conda install for pinocchio and source installation for crocoddyl.
While running the code I am receiving the following core dumped message:

`python: /opt/openrobots/include/eigenpy/numpy-map.hpp:160: static eigenpy::NumpyMapTraits<MatType, InputScalar, AlignmentValue, Stride, true>::EigenMap eigenpy::NumpyMapTraits<MatType, InputScalar, AlignmentValue, Stride, true>::mapImpl(PyArrayObject*, bool) [with MatType = Eigen::Matrix<double, -1, 1>; InputScalar = double; int AlignmentValue = 0; Stride = Eigen::InnerStride<>; eigenpy::NumpyMapTraits<MatType, InputScalar, AlignmentValue, Stride, true>::EigenMap = Eigen::Map<Eigen::Matrix<double, -1, 1>, 0, Eigen::InnerStride<> >; PyArrayObject = tagPyArrayObject]: Assertion ``(PyArray_DIMS(pyArray)[rowMajor] < INT_MAX) && (PyArray_STRIDE(pyArray, rowMajor))' failed.
`
`Aborted (core dumped)`

Kindly let me know

jcarpent, 2022-07-04:
You are mixing conda and the installation from apt via robotpkg.
Please unset your env variables related to robotpkg and only use conda. I guess you can also uninstall all the packages that you've installed via robotpkg and which are related to crocoddyl and Pinocchio.

spykspeigel, 2022-07-05:
@jcarpent 
I am working in a conda env with python 3.8.13 and with source install of pinocchio, example-robot-data, eigepy, hpp-fcl, crocoddyl

I tried conda install of all the above libraries except (crocoddyl) but things didn't work. So, the only thing that is working for now is the source installation of all the aobve metioned packages.
But the runtime of the examples like arm_manipulation is really bad. I never experienced such runtime like for 11 iterations its taking around 30 seconds. 

Do you know what could be causing the issue?

Is there any preferred installation method for doing source installation for example_robot_data and crocoddyl?

Regards,
Saroj

wxmerkt, 2022-07-05:
Hi Saroj, if you are using a source install - what did you set the CMAKE_BUILD_TYPE to? If you didn't set it, can you try rebuilding with `-DCMAKE_BUILD_TYPE=Release`?

spykspeigel, 2022-07-05:
Hi @wxmerkt, I had set it to Release during the installation. But still facing the issues.

jcarpent, 2022-07-05:
With such timings, it really means that you are not compiling in Release mode. It is easy to not set it. Why to not directly install croco using conda?

nim65s, 2022-07-05:
It must be done at configure time, then rebuild, then install, instead of only set it at install time

cmastalli, 2022-07-05:
I also suggest you to clean up the cmake files, e.g.,
```bash
cd build
rm -rf .
cmake ../
make install
```

wxmerkt, 2022-07-06:
Can you check how Pinocchio is configured? It's possible that WITH_COLLISION_SUPPORT is OFF, when it should be ON.

spykspeigel, 2022-07-06:
@wxmerkt It is set to True
`IF(BUILD_WITH_COLLISION_SUPPORT)
  SET(BUILD_WITH_HPP_FCL_SUPPORT TRUE)
ENDIF()
`

cmastalli, 2022-07-06:
@spykspeigel I suggest you to clean up again Pinocchio and enable the `BUILD_UTILS` and `WITH_COLLISION_SUPPORT`.
If you still have issue, then it would be better to create an issue in Pinocchio.
For now, I consider that we have provided support for your installation issue.

closed 2022-07-06 by cmastalli

Decembereye, 2026-01-01:
@spykspeigel  I also met this error.I wonder how the solution u take to solve it?