loco-3d/crocoddyl #1088: What are activations and residuals?

BolunDai0216, 2022-09-09:


nmansard, 2022-09-11:
Cost function are functions from state and control to scalars: l: x,u -> l(x,u) \in R. This is the basis API and you can build your own cost function by inheriting from CostModelAbstract. Yet, for convenience, most of the common cost functions are built under the following form:
l(x,u) = a(r(x,u)), where a: r->a(r) \in R is named the activation, and r: x,u->r(x,u) \in R^nr  is named the residual. a(r) typically acts as a pseudo-norm, and the most popular one (and the default choice in many cases) is the square L2 norm: a(r) = 1/2 r^T*r. The residual is typically the difference between a state or control function (eg the hand position, the COM, etc) and a reference value, eg r(x,u) = com(q) - com_ref  where com(q) is the position of the center of mass and com_ref is a reference (constant) value. 

Finally, the class CostResidualModel is approximating the cost Hessian as:
L_xx = R_x^T Arr R_x   (and similarly for the uu and xu derivatives). This approximation is knwon as "Gauss" approximation, and you will find in the litterature a justification for it. Let's just say that it is a fair approximation, leading to good convergence property, with the advantage of avoiding the computation of kinematic second order derivatives. 

Again, this is not a limitaiton of Crocoddyl, the basic API of CostModelAstract gives full control to the developer about implementing the full derivatives. This choice is limited to the scope of the particular implementation of CostResidualModel .

olivier-stasse, 2022-09-11:
Dear @nmansard since May 2022, github markdown support MathJax.

A small example of your answer with MathJax rendering:
> Cost function are functions from state and control to scalars: \$l: x,u \rightarrow l(x,u) \in R\$. This is the basis API and you can build your own cost function by inheriting from CostModelAbstract. Yet, for convenience, most of the common cost functions are built under the following form: \$l(x,u) = a(r(x,u))\$, 

Just in case you did not know...

cmastalli, 2022-09-11:
Thanks for @nmansard and @olivier-stasse for the explanations.

Please also see the doxgen documentation. This is all well explained there.

closed 2022-09-11 by cmastalli

BolunDai0216, 2022-09-11:
> Cost function are functions from state and control to scalars: l: x,u -> l(x,u) \in R. This is the basis API and you can build your own cost function by inheriting from CostModelAbstract. Yet, for [...]

Thank you so much for the answer!