loco-3d/crocoddyl #1089: Confusion with the cost function for the impulse model in the quadruped example

BolunDai0216, 2022-09-14:
Hi,

I was playing around with the quadruped example, and I was a bit confused with the cost function used within `createImpulseModel()`. From my understanding, the cost function has two parts one for swing leg tracking and one for all of the states reaching a target state. 

The swing leg tracking cost seems to be tracking the swing leg position at the knot point right before the impact, which is not on the ground. Would it be possible to help me understand why to use this position as the tracking target instead of the position when the swing leg contacts the ground?

Another confusion I have is with the state targets. It seems like the state targets are the default states, which would not match the states after a leg has moved:

```python
stateResidual = crocoddyl.ResidualModelState(self.state, self.rmodel.defaultState, 0)
stateActivation = crocoddyl.ActivationModelWeightedQuad(stateWeights**2)
stateReg = crocoddyl.CostModelResidual(self.state, stateActivation, stateResidual)
costModel.addCost("stateReg", stateReg, 1e1)
```

Why use the default state as the state target? Would the low weighting make this cost nonrelevant compared to the swing foot cost (whose weight is `1e7`)?

Thanks in advance!

cmastalli, 2022-09-14:
> Hi,
>
> I was playing around with the quadruped example, and I was a bit confused with the cost function used within `createImpulseModel()`. From my understanding, the cost function has two parts one [...]

Hi  @BolunDai0216,

Thank you for showing your interests in our software.

Before answering your question, I would like to tell you that the purpose of the examples are: (1) to let other deep more on how to formulate problems with Crocoddyl and (2) to track any development and its impact (sort of for debugging future developments). In short, they do not aim to be deploy on a robot.

Said that, we should define desired footstep placement at the impulse instant. If so, this is just tiny issue in the formulation of our example. But consider that we are defining the ground given the initial position. This is just trick for having a simple code.

On your second confusion, the state cost acts as a regularization term. It helps us to well pose the optimal control problem, and practically speaking is sort of hint to the solver. This represents a regularization cost as it plays a very small role when you compare with the other cost functions.

So, the answer of your last question is YES. That is exactly the idea as you can see now.

I hope this answer your questions. If so, could you close this issue?

BolunDai0216, 2022-09-14:
Thanks for the quick response, that answers my question!

closed 2022-09-14 by BolunDai0216