loco-3d/crocoddyl #1090: Compatibility with Numpy 1.23.0

nim65s, 2022-09-24:
Hi,

In the current devel, we have many instances on `np.asscalar`, which was deprecated in Numpy 1.16 and removed in 1.23:
https://github.com/numpy/numpy/pull/20414

It shouldn't be too hard to fix this, even for a newcomer :)

wxmerkt, 2022-09-26:
Hi @nim65s 
I fixed this both for the `devel` branch (but didn't make a PR due to conflicts with #1080), as well as for the preview-v2 which has been merged. As thus, it will be fixed as soon as as we merge #1080. Merging a fix to devel before only increases overhead/rebase conflicts

nim65s, 2022-09-26:
Thanks :) 
I'll need to get a patch with it for v1.9.0 ; do you have the commit that works on devel pushed somewhere ?

wxmerkt, 2022-09-26:
Here: https://github.com/wxmerkt/crocoddyl/commit/6901df512046f7fc05238da3af9faed6e188bd3a 

But I don't think we should commit it to devel and instead get v2 merged in

nim65s, 2022-09-26:
Applied in https://aur.archlinux.org/cgit/aur.git/commit/?h=crocoddyl&id=be2d5a5a3e86b0ba003cf6bcae18f0e5ab465e56, thanks :)

closed 2023-07-28 by cmastalli