loco-3d/crocoddyl #1108: State regularization

ThorbenPultke, 2022-12-02:
Hey once again! 

I am currently optimizing and extending my MPC for robot manipulators for my master's thesis. Right now I am bending my head around how I could possibly achieve state-regularization, but only for e.g. the joint-velocity of the state. From my point of view penalizing joint positions (with reference q_ref = 0) if you want to reach a target pose (e.g. for pick and place) is not what you would want. What I do want however is to penalize joint velocity, especially in the terminal node.

There is this Residual Model: "ResidualModelState", but this takes the whole state vector. Is there a convenient way to achieve regularization only for a specific part of the state vector with already implemented residual models? I am aware that by introducing a self-written residual model this problem is quite easy to solve, but right know I am bounded to python. Writing an own class in python would result in major computation time increasement, which is part of my evaluation criteria.

Thanks yet again for your help in advance!

Best Regards
Thorben

cmastalli, 2022-12-02:
You need to define the weights equal to zero for the parts of the state vector that you're not interested to regularise. This can be done via an activation model. That's it!

Just note that the residual of the state vector is always in the tangent space, therefore its dimension is `2*nv`.

ThorbenPultke, 2022-12-02:
Ah yes, of course! For those interested with similiar use-case: I just used ActivationModelWeightedQuad with a weight-array containing 0.0 for all position states and 1.0 for all velocity states, just like @cmastalli suggested. Thanks a lot!

closed 2022-12-02 by ThorbenPultke