loco-3d/crocoddyl #1109: FDDP for MPC and Implementation of a custom loss function

Flo-Wo, 2023-01-06:
Hi all,

first of all, thank you very much for developing such an incredible package. It really amazes me, how efficiently your FDDP solver is able to
find optimal and impressive movements. 

Currently I am working on a small project where we try to use your FDDP solver in an MPC
setting and add a custom loss function. It would be really helpful, if you could clarify some
questions/possible misunderstandings from my side.

1. Using FDDP in an MPC setting: 

Right now, we are still prototyping and therefore use the python bindings of your project (big thanks for
providing those!). For several reasons we are using mujoco as our internal simulator. To speed up calculations, 
we start by solving a task-space OC problem using the FDDP solver and then use a MPC with the FDDP solver to follow
the precomputed state-space trajectory.
While the FDDP solver can provide optimal trajectories (we were surprised by how aggressively close they 
are to the optimal expected behavior), our MPC implementing has problems with 
far-sighted enough time-horizons to follow the (optimal) trajectory while being executable in near real-time. 
I wonder if I implemented something wrong or misunderstood parts of crocoddyl. Here are the main 
parts of our implementation, it would be really helpful if you could give me some brief feedback on whether I made
a major implementation mistake. If the code below seems to be correct, then we have to come up with
different solution ideas.

```python
def mpc(...):
    state = np.concatenate((mujoco_data.qpos, mujoco_data.qvel))

    active_models = int_cost_models[:mpc_horizon]
    problem = crocoddyl.ShootingProblem(state, active_models, terminal_model)

    first_call = True

    num_total = len(int_cost_models)
    for idx, next_active_model in enumerate(int_cost_models[mpc_horizon:]):

        problem.x0 = state

        # create solver for this OC problem
        ddp = crocoddyl.SolverBoxFDDP(problem)

        # ==== LOGGING FOR DEBUGGING ====
        if idx % 100 == 0 and show_logs:
            print("\nINDEX: {}/{}".format(idx, num_total))
            log = crocoddyl.CallbackLogger()
            ddp.setCallbacks([log, crocoddyl.CallbackVerbose()])

        # check for the first time the MPC is called --> static warmstart and
        # we have to define the warmstart for the next calls
        if first_call:
            if start_with_static_warm and not custom_warmstart:
                logging.info("MPC: used a static warmstart for the first step.")
                x0 = np.concatenate(
                    (mujoco_data.qpos, np.zeros(np.shape(mujoco_data.qvel)))
                )
                custom_warmstart.set_warm(
                    [x0] * (problem.T + 1), problem.quasiStatic([x0] * problem.T), True
                )
            # only perform this step once
            first_call = False

        # solve the problem
        warm_x, warm_u, is_feasible = custom_warmstart.get_warm()
        solved = ddp.solve(warm_x, warm_u, solver_max_iter, is_feasible)

        # set the parameters for the next warmstart
        custom_warmstart.set_warm(ddp.xs, ddp.us, solved)

        # apply the first torque to the robot
        torque = ddp.us[0]

        mujoco_data.ctrl[:num_motors] = torque[:num_motors]  # apply torque
        mujoco.mj_step(model, mujoco_data)
        # update the current state for the next solution
        mj_x = mujoco_data.qpos
        mj_dx = mujoco_data.qvel
        state = np.concatenate((mj_x, mj_dx))

        # append the next problem node to our Problem-Instance
        problem.circularAppend(next_active_model)
```

Do you have any tips or advice on what I should look out for when implementing an MPC based on crocoddyl?

2. Implementation of a custom Activation Function:

We want to add a barrier function (cf. [here](https://en.wikipedia.org/wiki/Barrier_function)) as an activation function. Currently the c++ part seems to work and I also added a python binding file. Maybe a really stupid question, but where can I find the python bindings after a local build (i.e. calling ``make install``) and is there a simple way to import them after the build? Here a tiny hint would be really helpful.
Furthermore, is it correct that currently only activation functions with a diagonal Hessian matrix are supported (by the base class)?

As I am new to crocoddyl, I am grateful for any suggestions and hints :) Thanks!

Again, thank you very much for your hard work and the amazing package!

cmastalli, 2023-01-09:
Hi @Flo-Wo,

We are delighted to hear that Crocoddyl is helping you.

Your first questions is not entirely clear to me. Are you asking us how to boost the computation time of your current MPC code? If so, there are a few points to mention:
  - Do not create a Box-FDDP solver every time in the loop. This allocates solver data every time, which adds unnecessary computation time.
  - Compiles Crocoddyl with multi-threading support and use the optimal number of threads (typically equals to the number of physical cores)
  - Compiltes Crocoddyl with vectorization support. This enables further code efficient, but requires to compile all Crocoddyl dependencies with "-march=native".
  - Do not pass the action model only when calling "circularAppend". This allocates action data every time.

We add all the Python bindings of activation functions here: https://github.com/loco-3d/crocoddyl/tree/master/bindings/python/crocoddyl/core/activations. This requires to forward declare them here: 
 - https://github.com/loco-3d/crocoddyl/blob/master/bindings/python/crocoddyl/core/core.cpp
 - https://github.com/loco-3d/crocoddyl/blob/master/bindings/python/crocoddyl/core/core.hpp
Note that you might need to clean up your build folder. Then, if you do everything right, you can find your activation class as other Crocoddyl ones. Do not hesitate to create a PR if you contribute in a new (yet relevant for others) activation class.

If these responses answered your questions, then please mark the issue as close.

cmastalli, 2023-01-11:
Closing this issue due to lack of activity.

closed 2023-01-11 by cmastalli

Flo-Wo, 2023-02-28:
Thank you very much for your quick answer and detailed information. Sorry for the delayed answer, I needed some time to check and implement all of your suggestions. Now everything works like a charm, thank you so much for this amazing package and the hard work you put into the development!

fukangl, 2024-01-31:
Hi Florian@Flo-Wo,

I am also using MuJoCo as the simulator, were you able to successfully bridge Crocoddyl with MuJoCo? If so, could you share some insights on how to accomplish this? Thank you very much!

Flo-Wo, 2024-02-02:
Hello @fukangl :) Sure, in our work (available [here](https://github.com/Flo-Wo/IP2-Pendulum-Acrobatics)) we used Crocoddyl for the planning algorithm and Mujoco as an external simulator to emulate the sim2real transfer. The main insights for me were the following:
- Both work together very well in general, after correctly adjusting the parameters like integration time, friction params, etc.
- Depending on whether you model friction effects or not, there might occur some challenges. For instance, we used a Custom Crocoddyl dynamics Model to incorporate damping and armature (no problem), but we could not find a good modeling of the coulomb friction to adequately fit the Mujoco implementation. Thus, our custom model did not include coulomb friction (see the ``.pdf`` file for a more detailed discussion)

If you have any specific questions or my points were not clear enough, feel free to contact me here, via mail or LinkedIn :) I would be very happy to help you out!

cmastalli, 2024-02-02:
One important remark is on contact modeling. Crocoddyl supports rigid contacts and Mujoco simulates soft ones. It is also worth mentioning that all contact models are wrong but some of them are useful.

Crocoddyl (or any optimal-control) solvers are "aggressive". Reasons for aggressive behaviors are often related to an inappropriate problem formulation. If someone is getting this issue, then I would advise you to start looking at your regularization weights. This indeed can help to mitigate aggressive actions under noisy velocities.