loco-3d/crocoddyl #1119: DifferentialActionModelFreeFwdDynamicsTpl does not update data.pinocchio

huweiATgithub, 2023-02-24:
In the following code snippet:
https://github.com/loco-3d/crocoddyl/blob/0500e398861564b6986d99206a1e0ccec0d66a33/include/crocoddyl/multibody/actions/free-fwddyn.hxx#L124-L127

The derivatives are stored (expect ddq_dtau=Minv) in `d->Fx` instead of `d->pinocchio`, which makes it impossible for Cost Models to re-use these derivatives.

Any suggestion on that? Is there a workaround, or should we do the computation of the derivatives in each cost model that needs it?

cmastalli, 2023-02-24:
Hi @huweiATgithub,

These derivatives cannot be stored in Pinocchio as they do not represent what you meant.

However, you can always access the individual terms (i.e., `Minv` and `dtau_du`) in your cost function via the respective collector.

closed 2023-02-24 by cmastalli

huweiATgithub, 2023-02-25:
Thank you.
I need also ddq_dq and ddq_dv, which can be stored in Pinocchio. (e.g. another overload version of computeABADerivatives)

cmastalli, 2023-02-25:
The ABA derivatives are already stored in Pinocchio data, which you can also access via the collector.

huweiATgithub, 2023-02-27:
I checked the data. It is not saved in Pinocchio data due to the overload of computeABADerivatives Crocoddyl is calling (they are stored in the reference passing in instead).