loco-3d/crocoddyl #1122: Python binding `Solver.solve()` inflexible argument handling and exception translation.

Danfoa, 2023-03-15:
Hi I recently stumbled into some problems using the `solve()` function of the python `Solver` class. Specifically when trying to provide warm start trajectories. 

### List but not numpy arrays accepted for `init_xs` and `init_us`: 
The function seems to accept only a `list` of numpy arrays, and not a numpy array containing the warm start information. Furthermore, the exception raised provides little information about the nature of the issue. 
>>>
        x0 = some_initial_state  # of shape (nx,)     :     nx = nq + nv + nu

        problem = crocoddyl.ShootingProblem(x0, trajNodes, terminalModel)
        solver = crocoddyl.SolverDDP(problem)

        init_xs = [x0] * (problem.T + 1)                # List of numpy states [x0, ..., xT+1]    |   x ∈ R^nx
        init_xs_np = np.array(init_xs)                  # Numpy of shape (problem.T + 1, nx)    |    Same data

        solver.solve(init_xs=init_xs, init_us=[])     # No exception  
        solver.solve(init_xs=init_xs, init_us=[])     # Exception `Boost.Python.ArgumentError`

(The same issue occurs with init_xs). The exception raised is the following:
```
Boost.Python.ArgumentError: Python argument types in
    SolverDDP.solve(SolverDDP)
did not match C++ signature:
    solve(crocoddyl::SolverDDP {lvalue} self)
    solve(crocoddyl::SolverDDP {lvalue} self, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_xs)
    solve(crocoddyl::SolverDDP {lvalue} self, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_xs, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_us)
    solve(crocoddyl::SolverDDP {lvalue} self, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_xs, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_us, unsigned long maxiter)
    solve(crocoddyl::SolverDDP {lvalue} self, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_xs, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_us, unsigned long maxiter, bool is_feasible)
    solve(crocoddyl::SolverDDP {lvalue} self, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_xs, std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > init_us, unsigned long maxiter, bool is_feasible, double init_reg)
python-BaseException
```
Since the function signature has Eigen matrices as arguments, one would think that we should pass numpy arrays, considering the use of eigenpy. 

I am pretty unfamiliar with `boost.python`, but if you give me some initial guidance I can help augment the usability of the python bindings. Therefore I have a couple of questions:

 - How can we enable the use of numpy.arrays (matrices) of shape `(T+1, nx)`.
 - Is it possible to do some type/shape checking in python before forwarding the call to boost bindings? such that we enable some flexibility in the types used to pass the warm start trajectory data in python? (e.g., the user passes a list of list/tuples/numpy.arrays or a `(nx, T+1)` array instead of the expected `(T+1, nx)`) such that we reshape the data and cast to the appropriate types for the bindings? Or we need to do this on the C++ side by overloading the exposed function"

cmastalli, 2023-03-15:
Hi @Danfoa,

Is there a reason why we need to support NumPy arrays in `solve`?

Our current API supports vectors that change dimensions across nodes. We have two examples where we use that:
  - formulating problems with impulse dynamics in which nu=0
  - formulating problems with inverse dynamics as the control input changes according to the number of contacts.

These are all applicable to control vectors, but I wouldn't eliminate the possibility of different state dimensions in the future. For instance, this would be useful if we wish to formulate OC problems with a combination of full- and reduced-order dynamics.

In all those cases the issue is that we will have entries that are useless in the matrix object. This brings more confusion and room to make mistakes that are challenging to debug. To be honest with you, I don't think we should support this, but I look forward to hearing your motivations. If we agree to doing so, then I can tell you how to enable it.

Below, I provide some extra comments on selected quotes.

---

> ### List but not numpy arrays accepted for `init_xs` and `init_us`:
> The function seems to accept only a `list` of numpy arrays, and not a numpy array containing the warm start information. Furthermore, the exception raised provides little information about the nature of the issue.

Unfortunately, I am not aware if we can modify the `Boost.Python.ArgumentError` message. However, if you go through the Python documentation, it should be more clear, e.g.:
```python
In [1]: solver.solve?
Docstring:
solve( (SolverFDDP)self [, (StdVec_VectorX)init_xs [, (StdVec_VectorX)init_us [, (int)maxiter [, (bool)is_feasible [, (float)init_reg]]]]]) -> bool :
    Compute the optimal trajectory xopt, uopt as lists of T+1 and T terms.

    From an initial guess init_xs,init_us (feasible or not), iterate
    over computeDirection and tryStep until stoppingCriteria is below
    threshold. It also describes the globalization strategy used
    during the numerical optimization.
    :param init_xs: initial guess for state trajectory with T+1 elements (default [])
    :param init_us: initial guess for control trajectory with T elements (default []).
    :param maxiter: maximum allowed number of iterations (default 100).
    :param is_feasible: true if the init_xs are obtained from integrating the init_us (rollout)
    (default False).
    :param init_reg: initial guess for the regularization value. Very low values are typical
                     used with very good guess points (default 1e-9).
    :returns the optimal trajectory xopt, uopt and a boolean that describes if convergence was reached.

    C++ signature :
        bool solve(crocoddyl::SolverFDDP {lvalue} [,std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > [,std::__1::vector<Eigen::Matrix<double, -1, 1, 0, -1, 1>, std::__1::allocator<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > [,unsigned long [,bool [,double]]]]])
Type:      method

```

---

> * Is it possible to do some type/shape checking in python before forwarding the call to boost bindings? such that we enable some flexibility in the types used to pass the warm start trajectory data [...]

I am not sure what you mean by "type/shape checking". Currently, the solvers check the dimension of each state and control provided in initialisation. For more details see here:

https://github.com/loco-3d/crocoddyl/blob/82ace0c283170b4a1745c2cee303846df32948c2/src/core/solver-base.cpp#L184-L189

and here

https://github.com/loco-3d/crocoddyl/blob/82ace0c283170b4a1745c2cee303846df32948c2/src/core/solver-base.cpp#L212-L219

It is possible to add more optional signatures, but this needs to be included in the C++ code first. However, the flexibility that you mention might bring more confusion and unnecessary mechanism inside Crocoddyl (which leads to inefficiencies). We believe it is better to have a single and simple API. I personally believe that is the user tasks to make these conversions, if they are needed.

Danfoa, 2023-03-23:
> Our current API supports vectors that change dimensions across nodes. We have two examples where we use that:

Oh, I did not know this. If this is the case, now I understand the motivation behind the list. And after a closer look at the stack trace the `std::vector` was already hinting at a list of numpy arrays. My bad. 

I had already found the dimension-checking routines but did not understand why the dim was verified at each time step. Now it makes sense.

Thanks for the clarification and for taking the time.

cmastalli, 2023-03-23:
Ok, it seems this issue can be closed.

closed 2023-03-23 by cmastalli