loco-3d/crocoddyl #1124: Trajectory optimization of a wheeled robot

beaupollard, 2023-03-23:
Hi development team, I am trying to use the crocoddyl python api to determine an optimal trajectory for a wheel/legged robot (something similar to ANYmal). To start I created a simple 4 wheeled cart, I applied a 1D contact model at each of the wheels and I wrote a custom actuation class which maps the control input to forward acceleration of cart. When I run SolverFDDP the cart drives to the desired location but the contact forces that I'm recording are negative on some of the wheels. Obviously the contact force shouldn't be negative but I'm a bit lost on what to try next. Does anyone have some suggestions?

cmastalli, 2023-03-23:
Hi @beaupollard,

Our contact model imposes holonomic constraints at the acceleration level. So it is perfectly reasonable to have negative or positive forces. What you need to do to avoid this is imposing friction-cone constraints. This is also important to guarantee that the wheel is not slipping.

Perhaps, you can help us by writing an example of this problem. I could provide more feedback on how to improve this example.

beaupollard, 2023-03-24:
[crocoddyl_problem_statement.pdf](https://github.com/loco-3d/crocoddyl/files/11061726/crocoddyl_problem_statement.pdf)

Here is a brief high level pdf description of the problem I want to solve and what I've tried so far. Please let me know if something is unclear or if more detail is needed

cmastalli, 2023-04-17:
> 

Hi @beaupollard 

I apologise for the late response, but we had several project deadlines. Your document doesn't formulate the problem accurately. Unfortunately, you want us to identify solutions to your research questions. However, this needs to be discussed with your supervisors and collaborators. 

A 6D contact model constrains motions in 6 dimensions, but your wheel might want to translate. Optimising over contacts is a challenging problem and we are not yet solving it. Please read our Crocoddyl paper for more details.

I will close this issue.

linglluo, 2024-01-20:
> Hi @beaupollard
> 
> I apologise for the late response, but we had several project deadlines. Your document doesn't formulate the problem accurately. Unfortunately, this needs to be discussed with your supervisors and collaborators as it is not our responsibility.
> 
> A 6D contact model constrains motions in 6 dimensions, but your wheels need to translate. Optimising over contacts is a challenging problem and we are not yet solving it. Please read our Crocoddyl paper for more details.
> 
> I will close this issue.

Hi @cmastalli 
Thank you for your reply. Is there any paper or documentation for calculating the analytical derivative of the contact frame acceleration da0_local_dx?