loco-3d/crocoddyl #1130: some errors happen when importing crocoddyl

candyriver, 2023-05-01:
I am new  here and the errors are:
 
`Python 3.10.10 (main, Mar 21 2023, 18:45:11) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import crocoddyl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/.local/lib/python3.10/site-packages/cmeel.prefix/lib/python3.10/site-packages/crocoddyl/__init__.py", line 1, in <module>
    from .libcrocoddyl_pywrap import *
ImportError: libboost_chrono.so.1.80.0: cannot open shared object file: No such file or directory
`

cmastalli, 2023-05-01:
Hi @candyriver,

This might be related to a corrupted installation procedure. Can you provide more details on how you have installed Crocoddyl? Please read again the installation instructions.

candyriver, 2023-05-01:
I just follow the installation instructions and run the following command in the terminal:

  pip install --user crocoddyl

cmastalli, 2023-05-01:
@nim65s is aware with `pip` installation details. He might need further details to understand your situation.

jcarpent, 2023-05-01:
@candyriver Conda might be your friend `conda install -c conda-forge croccodyl`.

candyriver, 2023-05-02:
> @candyriver Conda might be your friend `conda install -c conda-forge crocoddyl`.

Thanks for your reply！ But sadly it didn't help.

candyriver, 2023-05-02:
After testing with Python 3.6, the error no longer occurred. This suggests that the issue only affects Python 3.10. I carefully reviewed the installation instructions available on https://github.com/loco-3d/crocoddyl/tree/devel and noticed that in the section labeled "From Debian/Ubuntu packages, with robotpkg", it instructs to set the PYTHONPATH environment variable as follows:
`export PYTHONPATH=/opt/openrobots/lib/python3.10/site-packages:$PYTHONPATH`
However, upon running apt-cache search robotpkg-py3*, I was unable to find a version of crocoddyl that is compatible with Python 3.10.

nim65s, 2023-05-02:
@candyriver : we offer multiple installation options. From what I can see here, you tried 2 different:
1. wheel packages downloaded from pypi.org with eg. `pip install crocoddyl` → Those packages require python >= 3.7, and the crocoddyl one might not be working until #1128 is out
2. deb packages downloaded from robotpkg.org with eg `apt install robotpkg-py3*-crocoddyl` → Those packages require using the `python3` provided by your Debian / Ubuntu installation, so in your 18.04 case only 3.6 will work

candyriver, 2023-05-02:
@nim65s  Thanks for your attention! 
Although the OS system is Ubuntu 18.04, I am using the latest conda environment, so my Python version is 3.10. However, I am still unsure why 'pip install --user crocoddyl' or 'conda install -c conda-forge crocoddyl' did not work for me. It's possible that these methods rely on the system installation, which may explain the issue.

nim65s, 2023-05-02:
if you are in a conda environment, deb packages installed with `apt` can't work. 

Both pip and conda packages should work, and shouldn't rely on system stuff,  but at the moment I think pip wheels are broken for crocoddyl, I'll fix this asap.

cmastalli, 2023-05-03:
Closing this issue as @candyriver find a way to install Crocoddyl. Thanks for providing support, @nim65s and @jcarpent!

closed 2023-05-03 by cmastalli