loco-3d/crocoddyl #1145: Obstacle avoidance utilizing pairCollision

amirrazmjoo, 2023-06-16:
I'm currently working on planning a collision-free path for a point mass system, and while researching similar issues like #1097, I came across a solution called 'ResidualModelPairCollision' that seems promising.

To implement this, I've set up the following components:

```
obstacleAvoidance = crocoddyl.ResidualModelPairCollision(state,actuation.nu, collision_model,0,state.pinocchio.frames[eeIdx].parent)
ObstacleAvoidanceActivation = crocoddyl.ActivationModel2NormBarrier(3,0.1)
obstackeAvoidanceCost = crocoddyl.CostModelResidual(state, ObstacleAvoidanceActivation, obstacleAvoidance)
```
Now, here's the issue I'm facing: When I initialize the problem with a path that collides with an obstacle, the algorithm returns false without any iterations. I'm wondering if I'm using the components correctly. Just so you know, in my setup, the obstacle shapes are spheres.

I appreciate any guidance or suggestions you can provide. Thanks!

cmastalli, 2023-06-29:
Hi @amirrazmjoo!

This could happen in principle. Returning `False` means that the solver is not able to converge. Have you tried to reduce the penalisation weight?

Could you share a simplified script that setups this problem?

cmastalli, 2023-07-28:
Hi @amirrazmjoo! I am lacking bandwidth. I suggest you discuss this topic with your supervisor.

closed 2023-07-28 by cmastalli