loco-3d/crocoddyl #1148: Undefined Symbols for architecture arm64 on MacM1 when compiling custom action model

Flo-Wo, 2023-07-06:
Hi all,

thank you for providing such an amazing package. After we achieved outstanding results in simulation with your package, we want to transfer our results to the real system and thus need a C++ implementation.

After building all the dependencies from source and then building crocoddyl 2.0 from source (which worked like a charm), I receive the following error 

```shell
Undefined symbols for architecture arm64:
  "crocoddyl::Exception::Exception(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, char const*, char const*, int)", referenced from:
      crocoddyl::CostModelSumTpl<double>::calc(boost::shared_ptr<crocoddyl::CostDataSumTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&) in ActionModel-ca09b6.o
      crocoddyl::CostModelSumTpl<double>::calcDiff(boost::shared_ptr<crocoddyl::CostDataSumTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&) in ActionModel-ca09b6.o
      crocoddyl::DifferentialActionModelAbstractTpl<double>::quasiStatic(boost::shared_ptr<crocoddyl::DifferentialActionDataAbstractTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1>>, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, unsigned long, double) in ActionModel-ca09b6.o
  "crocoddyl::Exception::~Exception()", referenced from:
      crocoddyl::CostModelSumTpl<double>::calc(boost::shared_ptr<crocoddyl::CostDataSumTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&) in ActionModel-ca09b6.o
      crocoddyl::CostModelSumTpl<double>::calcDiff(boost::shared_ptr<crocoddyl::CostDataSumTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&) in ActionModel-ca09b6.o
      crocoddyl::DifferentialActionModelAbstractTpl<double>::quasiStatic(boost::shared_ptr<crocoddyl::DifferentialActionDataAbstractTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1>>, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, unsigned long, double) in ActionModel-ca09b6.o
  "typeinfo for crocoddyl::Exception", referenced from:
      crocoddyl::CostModelSumTpl<double>::calc(boost::shared_ptr<crocoddyl::CostDataSumTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&) in ActionModel-ca09b6.o
      crocoddyl::CostModelSumTpl<double>::calcDiff(boost::shared_ptr<crocoddyl::CostDataSumTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&) in ActionModel-ca09b6.o
      crocoddyl::DifferentialActionModelAbstractTpl<double>::quasiStatic(boost::shared_ptr<crocoddyl::DifferentialActionDataAbstractTpl<double>> const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1>, 0, Eigen::InnerStride<1>>, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1>> const&, unsigned long, double) in ActionModel-ca09b6.o
ld: symbol(s) not found for architecture arm64
```

when I try to compile the following custom model:
```cpp
#include <crocoddyl/core/action-base.hpp>
#include <crocoddyl/core/actuation-base.hpp>
#include <crocoddyl/core/data-collector-base.hpp>

#include <crocoddyl/core/utils/exception.hpp>

#include <crocoddyl/core/diff-action-base.hpp>
#include <crocoddyl/core/costs/cost-sum.hpp>

#include <crocoddyl/multibody/states/multibody.hpp>

#include <pinocchio/algorithm/rnea-derivatives.hpp>
#include <pinocchio/algorithm/aba-derivatives.hpp>
#include <pinocchio/algorithm/compute-all-terms.hpp>
#include <pinocchio/algorithm/frames.hpp>
#include <pinocchio/algorithm/kinematics.hpp>
#include <pinocchio/algorithm/rnea.hpp>

#include "FrictionModel.hpp"
#include "ActionModel.hpp"

class DifferentialActionDataFrictionFwdDynamics: public crocoddyl::DifferentialActionDataAbstract {
    typedef crocoddyl::DifferentialActionDataAbstract Base;

public:
    DifferentialActionDataFrictionFwdDynamics(DifferentialActionModelFrictionFwdDynamics* const action_model):
        Base(action_model),
        pinocchio_data(pinocchio::DataTpl(action_model->pinocchio_)),
        multibody_data(
            &pinocchio_data, action_model->actuation_->createData(),
            boost::make_shared<crocoddyl::JointDataAbstract>(
                action_model->state_, action_model->actuation_, action_model->get_nu())),
        costs(action_model->costs_->createData(&multibody_data))
        // tmp_xstatic(model->get_state()->get_nx()) {
        // const std::size_t nv = model->get_state()->get_nv();
        // Fu.leftCols(nv).diagonal().setOnes();
        // multibody.joint->da_du.leftCols(nv).diagonal().setOnes();

        {
        // tmp_xstatic.setZero();
        costs->shareMemory(this);
    }

    pinocchio::DataTpl<Scalar> pinocchio_data;                  //!< Pinocchio data
    // DataCollectorJointActMultibodyTpl
[... 131 lines cut ...]
      ){

      }

boost::shared_ptr<crocoddyl::DifferentialActionDataAbstract> DifferentialActionModelFrictionFwdDynamics::createData() {
    return boost::allocate_shared<DifferentialActionDataFrictionFwdDynamics>(
        Eigen::aligned_allocator<DifferentialActionDataFrictionFwdDynamics>(), this
    );
}

int main(){
    return 0;
}
```

I use the following command to start the compiler:
```shell
g++ action_model/ActionModel.cpp -I"/opt/homebrew/include/eigen3" -I"/opt/homebrew/opt/boost@1.76/include" --std=c++17 -o ./action_model/action_model.out
```

I tried it with different version, e.g. c++20, but that did not work due to some error in the boost library which does not seem to be fixed yet.

Since I am not a C++ expert, I am really grateful for your hints.

Thank you very much in advance!

nim65s, 2023-07-07:
Hi,

I can't reproduce without FrictionModel.hpp & ActionModel.hpp. 
Could you either fix your cpp or include those files ?

Flo-Wo, 2023-07-07:
Ah yes, you are perfectly right. Sorry for the inconvenience and thanks for the quick response!

FrictionModel.hpp
```cpp
#ifndef  FRICTION_MODEL_HPP
#define FRICTION_MODEL_HPP

#include <Eigen/Dense>

class FrictionModel {
public:
    FrictionModel(const Eigen::Ref<const Eigen::VectorXd>& coulomb_friction,
                  const Eigen::Ref<const Eigen::VectorXd>& damping,
                  float coulomb_slope = 100.0
                  )
        : coulomb_friction_(coulomb_friction), damping_(damping), coulomb_slope_(coulomb_slope) {}

    // calculation methods
    Eigen::VectorXd calc(const Eigen::Ref<const Eigen::VectorXd>& v) const {
        return (-1) * (coulomb_friction_.array() * (coulomb_slope_ * v).array().tanh()).matrix() - v.cwiseProduct(damping_);
    }

    Eigen::MatrixXd calcDiff(const Eigen::Ref<const Eigen::VectorXd>& v) const {
        return ((-1) * coulomb_slope_ * (1 - (coulomb_slope_ * v).array().tanh().square())).matrix().asDiagonal() - damping_.asDiagonal();
    }

private:
    Eigen::VectorXd coulomb_friction_;
    Eigen::VectorXd damping_;
    float coulomb_slope_;
};
#endif  // FRICTION_MODEL_HPP
```

and 

ActionModel.hpp
```cpp
#ifndef ACTION_MODEL_HPP
#define ACTION_MODEL_HPP

#include <crocoddyl/core/action-base.hpp>
#include <crocoddyl/core/actuation-base.hpp>
#include <crocoddyl/core/data-collector-base.hpp>
#include <crocoddyl/core/utils/exception.hpp>

#include <crocoddyl/core/diff-action-base.hpp>
#include <crocoddyl/core/costs/cost-sum.hpp>

#include <crocoddyl/multibody/states/multibody.hpp>
#include <crocoddyl/multibody/data/multibody.hpp>

#include "FrictionModel.hpp"

class DifferentialActionModelFrictionFwdDynamics : public crocoddyl::DifferentialActionModelAbstract {
  public:
      DifferentialActionModelFrictionFwdDynamics(
          boost::shared_ptr<crocoddyl::StateMultibody> state,
          boost::shared_ptr<crocoddyl::ActuationModelAbstract> actuation,
          boost::shared_ptr<crocoddyl::CostModelSum> costs,
          const Eigen::VectorXd& armature = Eigen::VectorXd(),
          const Eigen::VectorXd& damping = Eigen::VectorXd(),
          const Eigen::VectorXd& coulomb_friction = Eigen::VectorXd(),
          const Eigen::VectorXi& grav_comp_idxs = Eigen::VectorXi()
      );

      void calc(const boost::shared_ptr<DifferentialActionDataAbstract>& data,
                const Eigen::Ref<const Eigen::VectorXd>& x,
                const Eigen::Ref<const Eigen::VectorXd>& u = Eigen::VectorXd());

      void calcDiff(const boost::shared_ptr<DifferentialActionDataAbstract>& data,
                    const Eigen::Ref<const Eigen::VectorXd>& x,
                    const Eigen::Ref<const Eigen::VectorXd>& u);

      // void quasiStatic(
      //   const boost::shared_ptr<DifferentialActionDataAbstract>& data,
      //   Eigen::Ref<VectorXs> u, const Eigen::Ref<const VectorXs>& x,
      //   const std::size_t maxIter, const float tol
      // );

      boost::shared_ptr<crocoddyl::DifferentialActionDataAbstract> createData();

      // make all attributes public
      boost::shared_ptr<crocoddyl::ActuationModelAbstract> actuation_;
      boost::shared_ptr<crocoddyl::StateMultibody> state_;
      boost::shared_ptr<crocoddyl::CostModelSum> costs_;
      // pinocchio part
      pinocchio::Model pinocchio_;
      // friction model
      Eigen::VectorXd armature_;
      Eigen::VectorXd damping_;
      Eigen::VectorXd coulomb_friction_;
      Eigen::VectorXi grav_comp_idxs_;
      FrictionModel friction_model_;
};
#endif  // ACTION_MODEL_HPP
```

cmastalli, 2023-07-28:
Closing this issue due to inactivity

closed 2023-07-28 by cmastalli