loco-3d/crocoddyl #1155: About setting up uneven terrain for bipedal walking

gaiyi7788, 2023-08-05:
Hi,
Thank you for sharing this library. It helps a lot the community.

I am working on a project on bipedal walking on uneven terrain. I would like to know if the Crocoddyl library supports features such as terrain settings to help me with this.

Look forward to your reply!

cmastalli, 2023-08-07:
Hi @gaiyi7788,

At the moment, Crocoddyl is not supporting these types of features. Although, some of them might be straightforward to implement.

As a reference, can you describe in more detail what you need?

cmastalli, 2023-08-07:
You can include obstacles using `ResidualModelPairCollision`. Unfortunately, we don't have an example of how to use it.

fukangl, 2024-01-31:
Hi Pengyu@gaiyi7788,
I am experiencing the same issue. Were you able to successfully bridge Crocoddyl with MuJoCo?