loco-3d/crocoddyl #1168: Two actuators but solved us are always the same pairwise

qodfathr, 2023-09-12:
I am sure this is just me being a novice with crocoddyl, so I appreciate any direction.

Starting with examples/notebooks/acrobot_urdf.ipynb, I wanted to see if I could actuate both joints and thereby solve for both u's.

I had to make some changes to the notebook:

```python
class AcrobotActuationModel(crocoddyl.ActuationModelAbstract):
    def __init__(self, state):
        nu = 2  # Control dimension
        crocoddyl.ActuationModelAbstract.__init__(self, state, nu=nu)

    def calc(self, data, x, u):
        # I observed that u[0] and u[1] are always the same value at this point
        assert len(data.tau) == 2
        # Map the control dimensions to the joint torque
        data.tau[0] = u[0] # joint at base of arm
        data.tau[1] = u[1] # joint at shoulder of arm

    def calcDiff(self, data, x, u):
        # Specify the actuation jacobian
        data.dtau_du[0] = 1
        data.dtau_du[1] = 1

...

# Add a cost on control
controlResidual = crocoddyl.ResidualModelControl(state, nu=actuationModel.nu)
bounds = crocoddyl.ActivationBounds(np.array([-1.0]*actuationModel.nu), np.array([1.0]*actuationModel.nu))

...

# Test the problem with a rollout
us = [0.01 * np.ones((actuationModel.nu,))] * T

...

# Display using meshcat
robot_display = crocoddyl.MeshcatDisplay(robot, -1, 1) # this change was just to get meshcat working

```

However, when I solve, I observed that the two u values were always the same; this graph is not in error: the u's are overlapping:

[image]

Can someone help me to understand what I did wrong? Greatly appreciated!!!

cmastalli, 2023-09-12:
You can easily actuate both joints via 'ActuationModelFull'. It is worth mentioning that you can also implement your own actuation model. For more details see its documentation.

Is this answer your question?

qodfathr, 2023-09-13:
Thanks, @cmastalli -- I'll look into `ActuationModelFull` and see if I can make it work. Appreciate the response!

closed 2023-09-13 by qodfathr