loco-3d/crocoddyl #1219: ROS_PACKAGE_PATH env variable is not set correctly after building crocoddyl with catkin build

aristotpap, 2024-02-01:
Hello,

Thank you for developing and maintaining **crocoddyl**. I am facing the following problem:

I have a catkin workspace and I want to build both **pinocchio** and **crocoddyl** from source using `catkin build`. I have cloned their dependencies too, namely **eigenpy** and **example-robot-data** into my workspace. I can build all packages correctly except  **crocoddyl**. When I try to build it, I don't face any errrors and the process finishes as expected. However, when I try to source my workspace using `source devel/setup.bash`, `roscd` and `rospack find` cannot find any package built in my workspace. After building crocoddyl the  `ROS_PACKAGE_PATH` variable is not set correctly, since  `echo` returns `/opt/ros/noetic/share`, even though my workspace is sourced.

If I set the environmental variable manually, `rospack` finds the packages as expected.
If I build any other package in my workspace after crocoddyl, `ROS_PACKAGE_PATH` is set correctly for it and its dependencies, but not for **crocoddyl**. 

One workaround to the problem, is to remove <depend></depend> statements, since the error seems to be build-tree related, when crocoddyl reinstalls its dependencies. Also, I tried to find a solution online and [this](https://answers.ros.org/question/264390/source-wont-change-the-package-path/) seems related, however I cannot tell what causes the problem.

I use ROS Noetic and catkin tools v0.9.0 with Python 3.8.10. Both **crocoddyl** and **pinocchio** are git cloned recursively. Also `CMAKE_PREFIX_PATH` is set as expected to `/opt/ros/noetic/`

My `catkin config` is the following:
[image]

cmastalli, 2024-02-01:
Hi @aristotpap,

Thanks for reporting this issue!

I have two questions for you
  1. Are you using the latest devel branch of Crocoddyl?
  2. Can you check if the `ros_ws/install/.catkin` files exist after compiling Crocoddyl?

Thanks,

aristotpap, 2024-02-01:
Hi @cmastalli ,

Thank you for replying promptly.

1. I have tried to build the `master` and `v2.0.2` branches. They both give me the same result.

2. I currently haven't setup an install space. I have only `devel`, as you can see in the following screenshot:

[image]

The corresponding `devel/.catkin` **before** building `crocoddyl` is  returning: `/home/akron/catkin_ws/src/eigenpy;/home/akron/catkin_ws/src/pinocchio;/home/akron/catkin_ws/src/example-robot-data`

After building `crocoddyl`, I am not able to find the `.catkin` file in `devel` folder. It seems that it is not created.

cmastalli, 2024-02-04:
@aristotpap -- could you test the PR in https://github.com/loco-3d/crocoddyl/pull/1220? This should sort out this issue.

Btw, I suggest you compile and install Crocoddyl without catkin. It is much faster and more convenient to do so.

aristotpap, 2024-02-05:
Hello @cmastalli,

I tried to build both PR in #1220  and **devel branch** (since I saw that you merged it) and I was not able to build it successfully. The errors though seem unrelated. [crocoddyl_build_logs.txt](https://github.com/loco-3d/crocoddyl/files/14166275/crocoddyl_build_logs.txt)
These are the errors I am getting:
[image]

Finally, I tried to `catkin build` by copying the `CMakeLists.txt` and `package.xml` from your PR #1220 to master branch. It built successfully, but the problem with the `ROS_PACKAGE_PATH` still persists. 

Note that during the weekend I was able to make it work by manually creating the `.catkin` file that you pointed out. But this is not a long term solution.

cmastalli, 2024-02-05:
Hi, @aristotpap!

The compilation error you reported is fixed in https://github.com/loco-3d/crocoddyl/pull/1221. You can see more details in that PR.

I'll merge this PR after passing all jobs (I hope this afternoon). However, I suggest trying this PR and reporting back.

When I compiled the latest changes, I could see the correct stored information inside `ros_ws/devel/.catkin`. Is it the same for you? I ask this I believe the issue with `ROS_PACKAGE_PATH` is there.

aristotpap, 2024-02-06:
Hello @cmastalli,

I tried building **devel** branch and after PR #1221 it was built successfully. However, the issue remains. Whenever I try to build **crocoddyl**, or state it as dependency in `package.xml`, the `ros_ws/devel/.catkin` file is not even created. If I manually create and populate it, the packages work fine.

cmastalli, 2024-02-06:
I am surprised as I couldn't replicate this issue in ROS 1. Is this happening in ROS 2?

aristotpap, 2024-02-06:
ROS 1 Noetic. It's an ANYbotics installation. I will try again with another machine and let you know. Thanks for the effort

closed 2025-01-30 by cmastalli