loco-3d/crocoddyl #1225: Inquiry on Incorporating Jerk Reduction into Cost Function Using Crocoddyl FDDP

sbin-github, 2024-02-19:
I'm a graduate student specializing in robotics. Please bear with me as my knowledge may not be as deep, and my question might lack some technical specificity.

I'm currently exploring the use of Crocoddyl FDDP in a robotics application. Given that the inputs are in the form of torques, I'm curious about the possibility of reducing jerk in the cost function. It seems logical to use the difference in torque values between previous knots for this purpose, but I'm uncertain if it's possible to create a differentiable cost that incorporates this aspect.

Could anyone share if there's a feasible method to incorporate jerk reduction into the cost terms within Crocoddyl FDDP? Alternatively, if there are practical approaches or formulas related to jerk that could be implemented, I'd greatly appreciate learning about them.

Thank you for your understanding and assistance.

cmastalli, 2024-02-19:
Hi @sbin-github,

Jerk is the rate of change in acceleration, not rate of change in torques as you seem to request. In any case, both rate of changes (acceleration or torques) are not part of our action decision variables. Therefore, your request is not possible as our solvers' assume a Markovian structure across action models. This is done to leverage efficiency.

An alternative would be to write your own dynamics and include jerks as decision variables. However, this is task goes beyond simple development and could be considered a research topic.

closed 2024-02-19 by cmastalli

sbin-github, 2024-02-20:
> Hi @sbin-github,
>
> Jerk is the rate of change in acceleration, not rate of change in torques as you seem to request. In any case, both rate of changes (acceleration or torques) are not part of our [...]

Thank you sincerely for your detailed response; it has been immensely helpful for advancing my project. I also want to apologize for any confusion caused by my initial question. Your guidance has clarified many aspects for me.