loco-3d/crocoddyl #1231: CMake couldn't find OpenMp

JianghanZHang, 2024-03-20:
## Bug description

Hello,

I encountered the following error when trying to build and install crocoddyl with multi-threading from source in a Conda environment. 

After running `cmake .. -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_MULTITHREADS=ON`
I got:
`CMake Error at cmake/find-external/OpenMP/FindOpenMP.cmake:222 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  cmake/find-external/OpenMP/FindOpenMP.cmake:549 (_openmp_get_flags)
  CMakeLists.txt:165 (find_package)`

## Reproduction steps

Upon creating a new Conda environment and activate

```
conda install python==3.10

conda install mim-solvers -c conda-forge

conda install cmake==3.23 pkg-config colcon-common-extensions -c conda-forge

conda install catkin_pkg empy lark -c conda-forge

conda install ncurses -c conda-forge

conda install websockets

conda install matplotlib

conda install -c conda-forge robot_descriptions

pip install pybullet

conda install pinocchio

conda install conda-forge::libgomp

conda install conda-forge::gcc

conda install conda-forge::gxx
```

### System

- OS: Ubuntu 22.04
- Crocoddyl version: 2.0.2
- Conda version: 24.1.2-0

It appears to me that CMake somehow couldn't find the installed OpenMp, but I have no clue how to fix it - even if I manually set CMAKE_MODULE_PATH, CMAKE_C_COMPILER, CMAKE_CPP_COMPILER, cmake still unable to find the OpenMp.

I appreciate your help.
Thanks!

cmastalli, 2024-03-20:
Hi @JianghanZHang!

Have you installed OpenMP? You could try Conda for that (see https://anaconda.org/conda-forge/openmp).

cmastalli, 2024-03-20:
Hi @JianghanZHang 

The issue appears when installing cmake v3.23. Please do:
```bash
conda remove cmake==3.23
conda install cmake
```

@nim65s is this expected to have a broken compilation with cmake v3.23?

cmastalli, 2024-03-20:
Should this issue moved to https://github.com/jrl-umi3218/jrl-cmakemodules/tree/59a8466a1143e2719b598b55d92a09d653628959?

closed 2024-09-07 by cmastalli