loco-3d/crocoddyl #1235: Once a backflip exceeds 180 degrees, anomalies may occur. Can you provide some ideas or a demo on how to solve this

dbdxnuliba, 2024-04-01:
## Bug description
Hello，When I use crocoddyl  trying to perform a backflip with a bipedal robot, if the target trajectory is set to 0-179 degrees, the backflip to 179 degrees is normal. However, once it exceeds 180 degrees, the backflip becomes abnormal, and the trajectory of the feet becomes abnormal. How can this problem be solved? Can you provide some ideas or a demo code on how to solve this.

- OS: [e.g. Ubuntu] ubuntu20.04

Many thanks

cmastalli, 2024-04-01:
Hi @dbdxnuliba,

Our examples are not designed to solve backflip or frontflip problems. We are planning to consider an example of this in the future. However, you'll need to be patient as this is not on our agenda in the next months. The reason is that we're planning to release this together with other changes as part of our ongoing research efforts.

In any case, Crocoddyl perfectly works with these types of examples.

I thank you for your patience and interest in our tool. Please consider contributing if Crocoddyl helps you in your work.

closed 2024-04-01 by cmastalli

dbdxnuliba, 2024-04-02:
is it a bug ,why it cannot exceed 180 degree