loco-3d/crocoddyl #1349: Assert fail in ActionModelLQRTpl::set_LQR

jorisv, 2025-03-03:
## Bug description

When building aligator with Crocoddyl 2.2.0 some unit tests are failing when build in [Debug mode](https://github.com/Simple-Robotics/aligator/actions/runs/13601475848/job/38027878389?pr=273).

We have the following assert message:

```
aligator-test-cpp-croc-problem: /home/jvaillan/dev/inria/prox/aligator/.pixi/envs/all/include/eigen3/Eigen/src/Cholesky/LLT.h:436: Eigen::LLT<_MatrixType, _UpLo>& Eigen::LLT<MatrixType, UpLo>::compute(const Eigen::EigenBase<OtherDerived>&) [with InputType = Eigen::Matrix<double, -1, -1>; _MatrixType = Eigen::Matrix<double, -1, -1>; int _UpLo = 1]: Assertion `a.rows()==a.cols()' failed.
```

That come from this code:

```cpp
  auto lqr_model = boost::make_shared<ActionModelLQR>(nx, nu);
  lqr_model->set_Luu(luu); // <=== assert here
```

When creating an ActionModelLQRTpl from the old constructor `ActionModelLQRTpl(const std::size_t nx, const std::size_t nu, const bool drift_free = true);` the `H_` matrix is initialized with 0 lines.

Then, `set_Luu` will call `set_LQR` that will call `Eigen::LLT` on `H_`.
We then trigger an Eigen assertion.

Also, since `set_LQR` is the only way to setup the `H_` matrix but the `Eigen::LLT` check is done before, it look like is not possible to run this code in debug anymore.

## Expected behavior

Since there is no API break between 2.2.0 and 2.1.0 we should still be build an `ActionModelLQRTpl` from the old constructor.
Also, we should be able to update the `H_` matrix without triggering an assert.

### Code

Build and run the following code in Debug mode with Crocoddyl 2.2.0:
```cpp
#include <crocoddyl/core/actions/lqr.hpp>

void main()
{
  using crocoddyl::ActionModelLQR;
  long nx = 4;
  long nu = 3;
  auto lqr_model = boost::make_shared<ActionModelLQR>(nx, nu);
  lqr_model->set_Luu(luu);
}
```

### System

- OS: Ubuntu 22.04
- Crocoddyl version: 2.2.0

cmastalli, 2025-03-04:
Hi @jorisv -- there is a typo in our `set_LQR` function, as we should perform the LLT decomposition of `L_`, not `H_`. I am happy to deal with this as soon as I finish merging the massive casting PR I have been working on for a month.

However, be aware that your LQR should have a cost Hessian that is positive semi-definite.

cmastalli, 2025-03-04:
I'll close this issues as it has been addressed in #1352.

closed 2025-03-04 by cmastalli

jorisv, 2025-03-05:
@cmastalli thank's !