loco-3d/crocoddyl #1398: Unexpected arg for Viewer

stephenmcc, 2025-05-30:
## Bug description

When trying to use meshcat or panda3d viewers, getting
"TypeError {MeshcatVisualizer/Panda3dVisualizer}.initViewer() got an unexpected keyword argument 'windowName'".

## Reproduction steps

1. Install v3.0.1 in conda environment (conda install crocoddyl -c conda-forge)
2. Install meshcat (conda install meshcat -c conda-forge)
3. Install panda3d_viewer (conda install panda3d_viewer -c conda-forge)
4. Install gepetto-viewer (conda install gepetto-viewer -c conda-forge)
5. Launch arm_example.py, with PINOCCHIO_VIEWER set to one of {meshcat, panda3d, gepetto}
6. For meshcat and panda3d, getting the TypeError below, for gepetto getting the AttributeError below

## Additional context

Traceback (most recent call last):
  File ".../arm_example.py", line 15, in <module>
    display = crocoddyl.GepettoDisplay(robot, cameraTF=cameraTF, floor=False)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../anaconda3/envs/mpc-data-generation/lib/python3.12/site-packages/crocoddyl/__init__.py", line 398, in __init__
    self._addRobot()
  File ".../anaconda3/envs/mpc-data-generation/lib/python3.12/site-packages/crocoddyl/__init__.py", line 460, in _addRobot
    self.robot.initViewer(windowName="crocoddyl", loadModel=False)
  File ".../anaconda3/envs/mpc-data-generation/lib/python3.12/site-packages/pinocchio/robot_wrapper.py", line 402, in initViewer
    self.viz.initViewer(*args, **kwargs)
TypeError: Panda3dVisualizer.initViewer() got an unexpected keyword argument 'windowName'

Traceback (most recent call last):
  File ".../mpc-data-generation/arm_example.py", line 15, in <module>
    display = crocoddyl.GepettoDisplay(robot, cameraTF=cameraTF, floor=False)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../anaconda3/envs/mpc-data-generation/lib/python3.12/site-packages/crocoddyl/__init__.py", line 398, in __init__
    self._addRobot()
  File ".../anaconda3/envs/mpc-data-generation/lib/python3.12/site-packages/crocoddyl/__init__.py", line 461, in _addRobot
    self.robot.loadViewerModel(rootNodeName="robot")
  File ".../anaconda3/envs/mpc-data-generation/lib/python3.12/site-packages/pinocchio/robot_wrapper.py", line 406, in loadViewerModel
    self.viz.loadViewerModel(*args, **kwargs)
  File ".../anaconda3/envs/mpc-data-generation/lib/python3.12/site-packages/pinocchio/visualize/gepetto_visualizer.py", line 181, in loadViewerModel
    gui = self.viewer.gui
          ^^^^^^^^^^^
AttributeError: 'GepettoVisualizer' object has no attribute 'viewer'

### System

- OS: Ubuntu 22.04
- Crocoddyl version: 3.0.1

cmastalli, 2025-06-03:
@nim65s  -- are you aware of what could be happening here?

nim65s, 2025-06-03:
I think you are right : not installing gepetto-viewer and installing panda triggers https://github.com/stack-of-tasks/pinocchio/pull/2419

nim65s, 2025-06-04:
ftr: akaik, gv is maintained :)